SWFSC/glider-lab #32: Create code for a synoptic view of glider and science data during a deployment

acossio, 2024-08-20:
It would be a nice thing to have to have something like this (https://marine.rutgers.edu/cool/data/gliders/deployments/deployment.php?deployment=bios_jack-20240812T1810) during a deployment to have a good long view of how the glider and science are sampling that looks better than the SFMC plots. 
This may be done by scraping the data off of the SFMC then providing plots. This will probably be a team effort.

acossio, 2024-09-19:
@flaimC is making great progress with this. His code is [here](https://github.com/flaimC/slocumRtDataVisTool/blob/main/rtGliderPlots.py)

flaimC, 2024-10-10:
The code works locally on my machine now to email out useful plots generated from raw binary files. I am working on getting to code to run upon VM reboot and am waiting to hear back about GCP CLI issues.

flaimC, 2024-12-02:
The script now runs ~4hrs on the GCP VM and either sends new data plots to the pilot team or a "no new data" message to me. 
All of the plots are saved on GCP and could be ported to a website of sorts for people to use. This will take some time due to the limitations of the VM using a hardened version of linux that does not easily support putting stuff on the internet.

acossio, 2025-05-07:
adding @smwoodman since I think he has taken this over. It's also being used in the semi-automated report.

smwoodman, 2025-07-29:
https://github.com/flaimC/slocumRtDataVisTool has been 'superceded' by https://github.com/SWFSC/slocumRtDataVisTool, so that Sam has write access to it.

This code runs as-is on deployments, and thus closing this issue. Future wish-list items:
- Update the code to read in the netCDF file created by [esdglider](https://github.com/SWFSC/esdglider). This would standardize processing (eg ensure profiles are calculated the same)
- Potentially move this class to the esdglider rt module

closed 2025-07-29 by smwoodman