SWFSC/glider-lab #39: Develop outline for practicing with glider data in GCP

smwoodman, 2024-08-26:
Make an 'actionable outline' for Jacob/Caleb/Tony to process data in GCP notebook. The purpose of this is to familiarize them with GCP services for running code on a Notebook instance, and for more generally running code in GCP using data stored in buckets.

smwoodman, 2024-08-26:
Sam items todo:
- [ ] Make 'test' bucket
- [ ] Provide some sample glider data
- [ ] Make task outline, send to all

smwoodman, 2024-08-26:
Task outline: https://docs.google.com/document/d/1rbFOVykNRP39pWdeGE9WO708R0eO0uxqw4CWnfe-pC8/edit?usp=sharing

HaileySchmidt-NOAA, 2024-10-01:
It seems like the link to the example config file in step 6.b.ii.2 is not working -  it says I do not have access.

smwoodman, 2024-10-01:
> It seems like the link to the example config file in step 6.b.ii.2 is not working - it says I do not have access.

Whoops, thanks @HaileySchmidt-NOAA 
Permissions should be updated now

HaileySchmidt-NOAA, 2024-10-01:
The permissions work now but when I download the file all I get is an empty .txt, could you make sure that the file uploaded is the correct one?

smwoodman, 2024-10-01:
Sigh, my bad. Correct file uploaded

smwoodman, 2025-03-14:
Not currently relevant, so closing. Can reopen if needed

closed 2025-03-14 by smwoodman