SWFSC/glider-lab #43: AMLR07 - glider restarting by itself

acossio, 2024-09-09:
AMLR07 has been restarting by itself during simulations. We need to figure out why it is restarting. Probably a short or loose wire. 
Glider support's response as of 9/4/24
The best guess I can give right now is there is some physical connection issue.  The evidence suggests that there is some momentary connection issue related to power.  

I have seen this before in one of our in-house gliders and was able to narrow it down to the MCBH connector (green plug).  I could cause the "restart" by wiggling the green plug.  This also is supported by the first instance of it restarting was when it was brought outside for the FCP (possibly jostled) and the second time was after 3 hours of simulation (possible vibration of the pump and pitch motor running overtime).

So my best recommendation would be to closely inspect all connections.  Specifically the following:

1.  White connectors between the main tray and science aft.  Check pins on both the male/female sides.
2.  2nd connection starboard side on the aft of the main tray.  This is labeled lithium and connects the green plug to the board.
3.  Wiggle the green plug itself when it is seated and the glider is powered on.  
4.  Check all connections at the BMS.
5.  Check for any loose ground wires.

I wish I had a better solution at this point but this is a gremlin.

acossio, 2024-12-30:
@flaimC  can you update this with what you have done with this glider please, so we have a record of it? Thank you

flaimC, 2024-12-31:
The issue with AMLR07 restarting of its own accord appears to have been inadvertently resolved. 

Here are the steps of how the issue was resolved:

- AMLR07 began restarting itself during a functional checkout prior to being deployed.
- We disassembled AMLR07 and checked for bent/broken pins and that all plugs on the flight computer board were seated. We did not find anything.
- We also tried several shorting plugs to ensure that the restarting issue was not due to a bad green plug. This did not resolve the issue.
- I went through with a multimeter and checked for continuity on all of the power wires (red and black). I also checked for continuity from the main board to the molex connectors for the larger plugs. I did not find any connections that were not continuous or continuous connections that did not make sense. I found one wire that I could not find the corresponding end of, but I suspect it is the female side of the shorting plug. I again checked that all connections on the main board were properly seated and plugged in.
- I then simulated AMLR07 outside and did not find it to restart.
- Additionally, we simulated AMLR07 in the ballast tank with a narrow shadowgraph for 4 days with no restarts.

I think I inadvertently fixed a loose connection while probing plugs and wires with the multimeter and this fixed the restarting issue.

My conclusion is that the restarting was due to a loose power wire connection going to the flight computer, causing the glider to randomly restart. To further conclude that AMLR07 is deployment-ready, I would like to deploy it off of San Diego for a short mission to see if it resets. This deployment can also be a shadowgraph test.

acossio, 2025-01-17:
Caleb noted that AMLR07 restarted again by itself when testing the glider. I think we try one more time investigating with @HaileySchmidt-NOAA while the glider is open and on. @acossio or @jgwalsh can assist with the prep of the glider to be able to turn on while it is open.

acossio, 2025-03-20:
This will have to be sent back to TWR when we can. We were not able to fix the issue.