SWFSC/glider-lab #45: Hefring Ocean scout glider data pipeline

acossio, 2024-09-13:
We should be getting two Ocean scout gliders in the near future. We have some sample data from the glider [here](https://drive.google.com/drive/folders/1XG3bSEU_9bIU1XZoBhgai_sifNuFhSaL) and some sample PAM data. Let's see if we can create similar products that we do with the Slocum glider data.

HaileySchmidt-NOAA, 2025-01-31:
**Information as of 1/30/25**

The raw files that come off of the Oceanscout (in the tarball) are:
1) Binaries with a proprietary .hfrng extension. It appears that each "category" of data generates its own series of files with unique prefixes that are mostly self-explanatory: ahrsData, ctdData, gpsData, IMR_01, navSensorData, pamData, and platformData.
2) CSV files for VerticalRatePID, PitchPID, and HeadingPID. These appear to track target values, actual values, motor positions, motor commands, and the values for the PID coefficients (which change dynamically?) Annoyingly, there are several header breaks inserted into the file, maybe generated every surfacing or some other interval, that will make this data slightly more difficult to read into a script. There are also no units.
3) kml track files: unsure if these are produced by the glider or if they were generated post-deployment by Hefring.
4) Logs, two kinds. The message_Log appears to be a plain text file of all the commands sent by the glider during the deployment. The lcmLog files are a weird mix of ascii and binary data, and I'm unsure what they're supposed to contain.

Hefring also provided us with NetCDF files and MATLAB .mat files, but as of now we are unsure how these were generated. Still waiting on a response from Hefring about this (emails sent 1/14/25, 1/24/25, 1/29/25).
- The provided NetCDF files are somehow corrupted and cannot be opened, so still unsure if the format will be compatible with our existing pipelines. 
- The .mat files are able to be opened in both MATLAB and Python and are standard structs.

The files are separated into science data (CTD_001.mat, CTD_001.nc) and flight data (GNSS_001.mat, GNSS_001.nc) which have different numbers of samples, so therefore most likely different sampling times (similar to Slocums).

The variables contained in file each are as follows:
- CTD Data: Time, TimeMatlab, Pressure, Depth, Temperature, Conductivity, Salinity, Density, ProcessorTime, Units, Comments
- GNSS Data: Time, Status_Date, Status_Time, Status_Resolved, Status_Declination, Status_Fix, Status_Differential_Corrections, Status_Power_Save, Status_Heading, Status_Carrier_Solution_Ambiguities, Status_Confirmed_Available, Status_Confirmed_Date, Status_Confirmed_Time, TimeMatlab, Accuracy, Satellites, Longitude, Latitude, HeightAboveEllipsoid, HeightAboveSeaLevel, HorizontalAccuracy, VerticalAccuracy, VelNorth, VelEast, VelDown, GroundSpeed_2D_, SpeedAccuracy, DilutionofPrecision, MotionHead, HeadAccuracy, VehicleHead_2D_, MagneticDeclination, MagneticDelinationAccuracy, Units, Comments
 
"Units" and "Comments" are structs that associate a unit for each measurement and describe some, but not all, of the variables. Although most of the variables are self-explanatory, there are a few that aren't clear and are not expanded on in the comments.

Finally, after speaking with Tony and Caleb, we have identified some data that we feel is important that is not currently in any of the Hefring data. It's unclear if these are able to be recorded but were not included in the output, if they're reported through the piloting terminal, or if the Oceanscout simply doesn't measure these values. These are:
- Battery info: voltage, current, total amp-hours used, status (this feels very important)
- Internal pressure/vacuum
- Leak detection/internal humidity
-  Altitude (there is an altimeter on the glider so we should be able to access this data)
- Buoyancy info: battery position, oil volume
- Pitch and roll (these are in the csv files but not the other data files - these are extremely important for our real-time monitoring)

Will continue to pester Freda and Becky at Hefring for a response to our questions.

acossio, 2025-05-29:
Email sent to Hefring after training (March 19, 2025)
- [ ] 1. Can you please provide calibration sheets for the hydrophones on the gliders? I can provide the SN#s if you need them.
- [ ] 2. Does the altimeter have a calibration sheet as well? If so, can we have a copy of that also?
- [ ] 3. Can we get a copy of the SDS sheet for the batteries? This way we can ship them to NE Fisheries team along with their glider?
- [ ] 4. We were able to view the full data set in Ocean Contour but it wasn't clear how we could export the data from Ocean Contour into a format that we can process with code. How do we get the data out of Ocean Contour as either a NC, CSV, or some other compatible file(s) that we could work with in R,Python, or Matlab?  Do we need to have an Ocean Contour license to do this?
- [ ] 5. We have a contract with DoD for RUDICS SIM cards. I know Jenny said that the OceanScouts don't use SIM cards. Do you know how we can connect our DoD account to the OceanScouts? What info would we need from you or what info would you need from them to match so we can use our account? We'll probably want to do this before deploying since we can't spend any money at the moment but we can use the contract that we have in place. 
- [ ] 6. Is there a way to add additional emails to the notifications from the glider? Currently, it is set to my work email and I am currently not allowed to view my work email while not at work. This will be an issue while piloting overnight if I get a notification. 

3. We did get a copy of the sds sheet.
5. We can't use the RUDICS cards. See other issue. 
6.  Currently no. They want us to use the app instead.