SWFSC/glider-lab #48: Add variables to NetCDF files

jgwalsh, 2024-09-19:
Can we add the variables "time" and "m_tot_num_inflections" to the engineering NetCDF files?

smwoodman, 2024-09-19:
Thanks @jgwalsh!

Will do re `m_tot_num_inflections`.

Time is already part of the netCSF file. It is a coordinate rather than a variable, so maybe it just needs to be accessed in a different way?

jgwalsh, 2024-09-19:
Yep - I figured out that "time" is just the same and is accessed the same
way (probably should have tested that before I created the issue!). Thanks
Sam!

smwoodman, 2024-09-26:
Added in latest [glider-utils](https://github.com/SWFSC/glider-utils) commit, so closing this issue. 

But see https://github.com/us-amlr/glider-lab/issues/50 for up-to-date info about what deployments have been processed in this new structure