SWFSC/glider-lab #49: New deployment project name(s)?

smwoodman, 2024-09-20:
Ask Christian/Heidi if we need new glider deployment project names. 

Currently we have FREEBYRD (Antarctic), REFOCUS (REFOCUS-specific deployments) and SANDIEGO (test deployments off of SD/Oceanside.

Do we want other project names, for instance for the george-20240907 Santa Barbara deployment, or for upcoming deployments off of Humboldt? Or do these fall under the scope of REFOCUS?

acossio, 2024-09-26:
My thoughts are that REFOCUS was a project that is now complete. There are two lease areas that we will likely be working in both in the future, Humboldt and Morro Bay. Maybe these are two new projects. Though Christian may just want a Wind project instead. I'm not sure about the collaboration projects such as the one with NASA. There may be a PAM deployment next year as well, possibly one-off Humboldt and one off of Hawaii. These I'm unsure of how they should be grouped.

acossio, 2024-09-27:
Sent email to Christian on 9/27/24

smwoodman, 2024-09-27:
Per CR: 

- ECOSWIM is the name of Wind-related deployments, eg off of Humboldt and Morro Bay.
- Move amlr08-20220513 to REFOCUS. REFOCUS is completed
- Happy with whatever for one-off deployments, such as the 2024 SB deployment of george. Current proposal is SOCALBIGHT

Todos:

- [ ] Move amlr08-20220513 to REFOCUS
- [ ] Make ECOSWIM project folder. Add to code as an acceptable value
- [ ] Make SOCALBIGHT (or equivalent) project folder. Add to code as an acceptable value
- [ ] Move george-20240907 to said folder
- [ ] Add project names to the database.? Sam to check if this is necessary

smwoodman, 2024-09-29:
Updated plan is to create a 'CCE' project for miscellaneous (eg, with collaborators) deployments in the CCE.

smwoodman, 2025-03-14:
Closing this issue given changes in glider program scope. Can revisit if needed

closed 2025-03-14 by smwoodman