SWFSC/glider-lab #51: Glider team Google email group

smwoodman, 2024-09-26:
It would be handy for sharing docs/GCP management to have an up to date glider team Google Group. 

Could update swfsc.aerd-gliders@noaa.gov, but likely will want a new, ESD-specific group.

Sam will ask Heidi/Christian

acossio, 2024-09-26:
I was thinking about this earlier this week. Either update the list or start a new one. The current list has George and Heidi and Christian. I'm not sure if they all need to be on the list. Maybe just the worker bees, like us. Would be good to ask Heidi/Christian.

smwoodman, 2024-09-27:
Ticket created: https://apps-st.fisheries.noaa.gov/jirasm/servicedesk/customer/portal/8/SWFSC-1791

smwoodman, 2024-09-27:
Group created: nmfs.swfsc.esd-gliders@noaa.gov

closed 2024-09-27 by smwoodman