SWFSC/glider-lab #60: calanus post-deployment tasks 202410

prtda, 2024-11-18:


jgwalsh, 2024-11-18:
- [ ] Open glider
- [ ] Remove aft end, set somewhere safe
- [ ] If using a card reader (fastest method), remove flight SD card. If downloading files over freewave, let's talk about how to use the high-speed feature
- [ ] Copy data files from SD card to computer
- [ ] Remove science computer SD card (should be visible in a slit in the aft end of the science bay after you remove the battery and the aft hull)
- [ ] Copy data files from SD card to computer
- [ ] Replace SD cards
- [ ] Seal up glider
- [ ] If o-rings seem dry, lube them up a little. They don't need to be replaced yet, as they'll be replaced before the next time the glider goes in the water
- [ ] Pull vacuum
- [ ] Turn glider on, check vacuum
- [ ] Deflate air bladder
- [ ] Retract oil
- [ ] Turn off glider
- [ ] For compressed data, uncompress .dcd and .ecd files (to convert them to .dbd and .ebd files) using the compexp.exe executable on the Datahost. This can be done from a terminal window, or Jen has R code that will do it, and Caleb may have Python code that will do it 
- [ ] Upload data (.dbd and .ebd files) to GCP (we can go over this)
- [ ] Later, once sure that all data files have been uploaded to GCP, delete data from SD cards by turning on glider, typing "loadmission lowpow.mi", then "dellog all". You will need to confirm that you want to delete all log files from both the flight and science computers. You can confirm that all files were deleted by going to the "logs" and "sentlogs" files on both computers. If "dellog all" doesn't work for all folders, you may have to manually navigate to each folder and delete the contents

prtda, 2024-11-26:
Note, we don't have a vacuum pump up here yet so I can't pull the vacuum, and thus cannot turn the glider on for now.

prtda, 2024-12-02:
I uploaded the compressed files to GCP, will try to un-compress them this week

acossio, 2025-01-17:
@prtda were you able to check if the optode was what failed or the cable of the optode failed?

prtda, 2025-01-17:
I can check next week, will need a refresher on the uart method

prtda, 2025-02-05:
Note, still need to delete old data. Waiting for the O.K.

acossio, 2025-02-06:
If data is uploaded to GCP, you are free to delete the data. Just double check that it is uploaded.

closed 2025-06-17 by acossio