SWFSC/glider-lab #62: get data off sig100 and send wet bay back to LJ

prtda, 2024-11-20:


prtda, 2024-11-26:
spending some time figuring out the best way to get data off of the sig100. Trying Nortek MIDAS, if that doesn't work I found the commands to print directly to a terminal.

prtda, 2024-12-02:
I think I got all the data off the sig, now to dig into it.

acossio, 2024-12-30:
@prtda sent this down to LJ. Tony looked into seeing if more than just the two days of data are recoverable. I followed these [instructions](https://drive.google.com/file/d/1IWTjwxIF4RjhVysI3Qt62J6eb2mphu5u/view?usp=drive_link). I tried image the memory card using a Linux machine. Then on a windows machine, I used the redimgcopy.exe that Nortek had given us in the AMLR Gliders -> Software -> Nortek Google drive. I had no luck as it said that "doesn't contain Nortek Edge partition (type 246)"
Looking back at my notes, I think that process was just for the ADCP, not the mini-echosounder. I'll try the instructions from [my notes](https://docs.google.com/document/d/11ih0V16ZixwDb9c7f-6vLbt-bbhiwrVagAG_7VGKJaU/edit?tab=t.0) next week since this was specifically for the mini-echosounder.

acossio, 2025-01-13:
The data was recovered and uploaded to the bucket amlr-upload-tmp-dev as calanus.ad2cp

closed 2025-01-13 by acossio