SWFSC/glider-lab #65: investigate failed ODO sensor on calanus

prtda, 2024-11-26:
- [ ] first step, uart to sensor to see if I can talk to the sensor. (we tried this while it was deployed with no luck. will try again on land, maybe while lightly pushing the cable around.)
- [ ] then will test continuity in the cable, especially with the same sharp bend in the cable as in deployment

someone else can let me know if you have other ideas.

prtda, 2024-11-26:
note, will need a vacuum pump before I can turn the glider on to check this with the uart method.

prtda, 2025-02-05:
Update! 

First we checked whether we could talk to the optode in the lab. For posterity, here are the steps for that:

- Turn on glider (and go into lab mode, and `loadmission lowpow.mi`)
- `consci`
- `cd config`
- make sure the optode block is uncommented in `proglets.dat` and note the info there
- talk to the optode, or attempt to: `uart j1 9600 c 2` (port name j1 baud rate 9600 and bit number 2 are dependent on the sensor)
- if lines of data start printing, good. if not, bad.

Nothing was printing. Eventually Hailey sent a new sensor and cable. First thing I tried was the new sensor. And immediately we got data printing. 

Surprisingly (or not), it was the sensor that failed. @acossio @jgwalsh  should I attempt to figure out why? Or have you already been down this road before?

Next, I need to mount the new optode sensor. I'd also recommend checking its accuracy, unless folks are already confident in that. We have the tools to be able to do so here at the CPH lab, except maybe the appropriate cable for talking to the sensor.

acossio, 2025-02-06:
Good news. Just to double check that it is the sensor, try changing the cable between sensors to see if you get the same results. We have prepaid service for 3 optodes so if it is bad, we'll just send it in to TWR for service and they can see what happened. I don't think we have the skills to troubleshoot more than checking the cables and sensor. Nice job!

prtda, 2025-02-06:
@acossio we checked the old optode again using both cables, and also the new optode using both cables. The issue is definitely the old optode. Let me know where to send it.

acossio, 2025-02-07:
We should be putting a small dab of DC4 on our connections before plugging them in. We haven't always done it. 
You can either send it down to the lab or bring it down with you in February.  Good to be definite that it is the sensor.

closed 2025-06-17 by acossio