SWFSC/glider-lab #70: Incorrect CDOM values for instruments serviced before Jan 13, 2023

acossio, 2025-01-13:
Seabird has sent an [announcement ](https://oceanobservatories.org/2024/12/sbs-issues-notice-for-certain-cdom-fluorometers/?utm_source=OOI+Newsletter&utm_campaign=b4fc2af9ea-EMAIL_CAMPAIGN_2024_08_24_03_18_COPY_01&utm_medium=email&utm_term=0_-7f91c5f51d-176569205) that states that there may be incorrect CDOM values if an instrument was serviced before Jan 13, 2023. We need to figure out how to apply a correction. Also new optical heads purchased or serviced between Jan 2021 and July 2023 may contain an out-of-tolerance UV LED. "This CDOM data is irretrievable from CDOM fluorometers built with out-of-tolerance UV LEDs." 

[Sea-Bird-Scientific-CDOM-Data-Notice.pdf](https://github.com/user-attachments/files/18401125/Sea-Bird-Scientific-CDOM-Data-Notice.pdf)
[Sea-Bird-Scientific-UV-LED-CDOM-Notice.pdf](https://github.com/user-attachments/files/18401124/Sea-Bird-Scientific-UV-LED-CDOM-Notice.pdf)

acossio, 2025-01-17:
Look for deployments that this affected.

acossio, 2025-01-27:
@HaileySchmidt-NOAA 
Would you mind working on going back through all the deployments (San Diego, Antarctica, etc) and seeing if there is bad CDOM data? Christian said that you can just plot a histogram of log10 CDOM and a histogram of log10 backscatter to tell.  (ie hist(log10(cdom(:)),100). Calanus had bad CDOM so you can use that as a reference.  Maybe create a document with all the histograms for easy comparison? 
Also, can you please make a note if there was acoustics (Nortek or AZFP) or a camera (glider cam or shadowgraph) on the glider during the deployment?  You might be able to use [Fleet Status Deployment tab ](https://docs.google.com/spreadsheets/d/1tB3QNKYx8qOYYS9QZotekBAx0y-_n2d-EZPjFFdYNuU/edit?gid=1505877661#gid=1505877661) for most of that. I'll pull the SNs of the Ecopucks for each deployment.

HaileySchmidt-NOAA, 2025-01-31:
Initial analysis:

I downloaded all of the NetCDF files that we have for past deployments (some deployments only had binaries or had corrupted NetCDF files). For each deployment, I calculated summary statistics on the CDOM data, deployment length, number of unique values, and "resolution" between data points. Each deployment has four plots: 

1. A histogram of the CDOM values with counts in log10
2. A histogram of log10(CDOM) values with counts in log10
3. A scatterplot of CDOM values across the deployment (x-axis is sample number)
4. A scatterplot of CDOM values by depth

In the second half of the script, I reloaded all of the deployments and roughly removed the outliers (by clipping values to [-5, 5] - CDOM shouldn't be negative, but a lot of the deployments had negative values, maybe due to a calibration issue/needing an offset, so I left those in). All of the statistics and plots are generated from this cleaned data.

The output of the script is here: [cdom_plots.pdf](https://github.com/user-attachments/files/18611084/cdom_plots.pdf)

I also went through the deployment database and associated each deployment with the serial number of the Eco PUCK that was on the glider. That data can be found here: [CDOM_Deployments_serials.pdf](https://github.com/user-attachments/files/18611088/CDOM_Deployments_serials.pdf)

Kourtney-Burger, 2026-05-07:
I finished updating the spreadsheet. I couldn't find autoexec files for almost all pre 2023 deployments but based on deployment and calibration dates I think it is safe to assume these used the original puck constants so the autoexecs should have been correct. 

Only amlr06-20220425 was deployed with a recalibrated puck, so that's the only missing autoexec that I am not confident about. 

**Next steps**
- [ ] script in the corrections for the wrong autoexec factors - corrected values listed in [table 4 here](https://docs.google.com/spreadsheets/d/1tB3QNKYx8qOYYS9QZotekBAx0y-_n2d-EZPjFFdYNuU/edit?pli=1&gid=952240415#gid=952240415)
- [ ] use the reference adjustment factor for deployments with incorrect cdom values 
> For the incorrect CDOM standard, we have determined a Reference Adjustment Factor (RAF) of 5.62 to apply to data: CDOM adjusted = 5.62 * CDOM

Kourtney-Burger, 2026-11-09:
All pucks that need to be serviced (due to out of tolerance UV lamps) have been sent in for calibration, waiting for 5704 and 6866 to be returned with sci bay services. There are 6 pucks last serviced in 2023 so those can be a priority to be serviced once we get on a good routine