SWFSC/glider-lab #75: AMLR03, AMLR04, AMLR06 Science Bays shipped to TWR

HaileySchmidt-NOAA, 2025-02-04:
Three science bays sent to TWR to remove IR filters from cameras, fix/replace CTD, repair scratch.

Bay 1: From AMLR03, originally from AMLR06, S/N 1341
Bay 2: From AMLR04, originally from AMLR02
Bay 3: From AMLR06, S/N 1546

acossio, 2025-02-07:
@flaimC  can you please write the RMA numbers in here for these items? Also please include them on [Fleet Status](https://docs.google.com/spreadsheets/d/1tB3QNKYx8qOYYS9QZotekBAx0y-_n2d-EZPjFFdYNuU/edit?gid=373748412#gid=373748412) in the RMA tab? Thank you

flaimC, 2025-02-08:
amlr02 (CRO-14440)
amlr03 (CRO-14441)
amlr06 (CRO-14442)

I think this is the correct pairing - Elise corrected the mismatch of science bays, but did not say if she reassigned the RMAs or changed the bays in the RMAs.

acossio, 2025-06-30:
We received all 3 science bays back. CRO 1442 came back with the scratch still on the sealing surfaces. After a call with TWR, they created a new CRO (CRO-15714) and is going to expedite the service.