SWFSC/glider-lab #79: Compile a List of Specific Questions for Hefring

HaileySchmidt-NOAA, 2025-02-20:
As we run into some areas of confusion and a lack of information from Hefring, it would be good to compile a list of specific questions to ask and areas where we need more clarification. For example, this would include variables missing from the sample data that we think are important (such as battery voltage), ambiguous variable names, login info, requests for manuals/SOPs, etc. We can decide later if these are questions we would like to send in an email or ask in-person during training.

acossio, 2025-03-20:
Questions were sent to Freda and Jenny on March 19, 2025. 
1. Can you please provide calibration sheets for the hydrophones on the gliders? I can provide the SN#s if you need them.
2. Does the altimeter have a calibration sheet as well? If so, can we have a copy of that also?
3. Can we get a copy of the SDS sheet for the batteries? This way we can ship them to NE Fisheries team along with their glider?
4. We were able to view the full data set in Ocean Contour but it wasn't clear how we could export the data from Ocean Contour into a format that we can process with code. How do we get the data out of Ocean Contour as either a NC, CSV, or some other compatible file(s) that we could work with in R,Python, or Matlab?  Do we need to have an Ocean Contour license to do this?
5. We have a contract with DoD for RUDICS SIM cards. I know Jenny said that the OceanScouts don't use SIM cards. Do you know how we can connect our DoD account to the OceanScouts? What info would we need from you or what info would you need from them to match so we can use our account? We'll probably want to do this before deploying since we can't spend any money at the moment but we can use the contract that we have in place. 
6. Is there a way to add additional emails to the notifications from the glider? Currently, it is set to my work email and I am currently not allowed to view my work email while not at work. This will be an issue while piloting overnight if I get a notification.

acossio, 2025-06-17:
This is all addressed in [issue 45](https://github.com/SWFSC/glider-lab/issues/45)

closed 2025-06-17 by acossio