SWFSC/glider-lab #81: Review calibration docs

smwoodman, 2025-03-13:
Hey @acossio - I moved up all of the calibration docs from the sql database into glider-lab. Because had a variety of naming patterns, I took the lazy way and made folders for the different instrument types (yes I know, the exact thing I said no to yesterday :)

Take a look through these to sanity check them, and make sure it doesn't look like any big chunks are missing?

acossio, 2025-03-20:
@smwoodman Most look good. We should have at least 12 Optode calibrations.  I renamed most of the files so they are consistent.

smwoodman, 2025-03-21:
@acossio you're totally right - there are a bunch more optode files that I scraped out of the database. Not sure why I missed them on pass one. 

I just pushed them up

smwoodman, 2025-03-21:
Hey Tony, one more thing. I uploaded all of the files that came out of the SQL Server database to https://drive.google.com/drive/folders/1627rIbGP7ehg0tvuAwaKFRrRK1vEwHhB?usp=drive_link

Take a look through these and double check that nothing else (especially from the na folder) should be up in the calibration-docs folder? The folders into which the files are grouped are the ‘file types’ from the database. The ‘na’ folder is files that did not have an associated file type

acossio, 2025-03-21:
Thanks. That looks much better with the optode cal sheets. I renamed the optode cals again. I also combined some files that were in two parts.

acossio, 2025-04-15:
@smwoodman  are we good to close this out?

smwoodman, 2025-04-15:
Good by me!

closed 2025-04-15 by smwoodman