SWFSC/glider-lab #83: Acoustics - move non-Echoview processing to Python

smwoodman, 2025-04-03:
Currently, Tony's acoustic processing involves using outputs from SOCIB, as well as intermediate products that he needs to create by hand from said SOCIB outputs. While Echoview will still be used to actually process the acoustic data, this issue is to shift all of the surrounding tasks to Python via pyglider/esdglider

- [ ] Sam creates dive and climb region evr files as part of esdglider.acoustics processing
- [ ] Tony tests various files in Echoview pipeline
- [ ] Get the Echoview output to the gridded NetCDF file

This will make this pipeline more robust/consistent, and let us make these gridded data available via ERDDAP

acossio, 2025-04-03:
Tony was using SOCIB processed files with Matlab and R code for the different files needed to process the glider acoustic data in Echoview. Having it all through the pyglider and Python will streamline the process.

smwoodman, 2025-04-04:
@acossio just emailed first draft of all files. Happy to walk through them together

smwoodman, 2025-04-04:
Request from Tony:  
For naming, can you name the gps, pitch, roll as "name.gps.csv", "name.pitch.csv." Echoview looks for the .gps.csv for the file name, not -. 

Also, regions have some funk, ie the profile identifier needs to be debugged

acossio, 2025-04-04:
At the end of the deployment there are still issues with the climb regions. Places where there are breaks in regions. Also, it is registering some of the surface time as climbs. Not all the time, but sometimes.

acossio, 2025-04-04:
It looks like whenever there is a "wobble" in the glider's trim, the regions are starting and stopping. I see this mostly in the climb regions but now that I am looking through the data, it is also on the dives, just not as much.

smwoodman, 2025-05-07:
An update: Sam has mostly figured out how to use the findProfiles function, so the above regions issues should be solved.

Tony is investigating differences in the number of profiles, among other things (https://github.com/SWFSC/glider-lab/issues/87)

The among other things includes exploring why the glider m_depth and depth_ctd are different by up to 30m at depth..

acossio, 2025-05-12:
Differences in the number of profiles seemed to be stemming from Tony's Matlab code where it was creating its own profiles based on profile direction and profile number. I changed the code to assume that odd integers were dives. I may have to make some checks on this, but it now has the same number of profiles as the Socib output.  Densities between the two processes are pretty close but not exact. 
NC
gm2_std =  101.2784
gm2_ste =    5.6528
mean_gm2 =   45.7913
biomassBS =   4.1212e+05

SOCIB
gm2_std =   97.4411
gm2_ste =    5.4386
mean_gm2 =   44.7576
biomassBS =   4.0282e+05

acossio, 2025-05-12:
Some Q's for @smwoodman  (more may be added)
- [ ] Do we know why the depth bins for the gridded data start at 0.5m [0.5, 5.5, 10,5, etc]? I don't think it really matters though. 
- [ ] Can we add oxygen concentration to the nc file?
- [ ] Currently, the acoustics goes down 100m below the glider. The revised Nortek should go down 200m. Do we want to increase the gridded data to go down to 1200? Currently it goes down to 1095.5m
- [ ] Profile_time_start and Profile_time_end are not times in the gridded data
- [ ] I need to speak with other acoustic people and see if just adding abc for acoustic output is good or if they would want other output, such as min, mean, max dB

I'm looking to append the gridded file that pyglider created. It should have 70kHz_abc and 120kHz_abc.

acossio, 2025-05-13:
I'm looking at what attributes as well as acoustic data should be included in the netcdf files that were created from pyglider. Doing a quick look, [echopype](https://echopype.readthedocs.io/en/latest/data-format-sonarnetcdf4.html) uses the [ICES Sonar-netcdf4 convention ver.1](https://ices-library.figshare.com/articles/report/The_SONAR-netCDF4_convention_for_sonar_data_Version_1_0/18624056?file=33870818) when possible.  Version 2 of the ICES netcdf4 is live in [Github](https://github.com/ices-publications/SONAR-netCDF4https://github.com/ices-publications/SONAR-netCDF4).
Most of this is geared more towards full data, not just gridded data. I'll speak with AST on 5/14 and discuss some of this to see how we should move forward.

smwoodman, 2025-06-06:
> * Do we know why the depth bins for the gridded data start at 0.5m [0.5, 5.5, 10,5, etc]? I don't think it really matters though.

Confirmed with pyglider authors that even though these are the depth bin values, the depth bins do go from 0-5m, 5-10m, etc. These values are labels; they were intended to be the midpoint of the bin, but there was a bug where the label calculation assumed 1m bins