SWFSC/glider-lab #90: Add firmware/software version to glider metadata

acossio, 2025-05-06:
Add firmware and/or software version for sensors that get updated
- [ ] glider
- [ ] AZFP
- [ ] Nortek mini echosounder
- [ ] Wispr
- [ ] DMON
- [ ] Shadowgraph
- [ ] Glidercam

Also for moorings
- [ ] ADCP-e
- [ ] CTD

@smwoodman I'll look, but maybe we can add a firmware version to the sensors in the database as well so it will be easier to track. We can also add this to the yaml files?  I'm open to ideas.

smwoodman, 2025-05-06:
Yes! I agree this is a very important thing to do. Because one device can have multiple versions, I'll need to add a new table, eg device_firmware. This way, we can have multiple entries for one device. 

What columns will this table need? Some that come to mind:
- Device_ID: to link to Device table
- firmware_version: a text string with the firmware version identifier
- firmware_date: the date firmware_version was installed

What else changes across the same device over time? Do any devices have both a software_version and a firmware_version.?

smwoodman, 2025-05-06:
Also, agreed this will be important to pull out into the yaml file. Once we add this table to the database, then I can update the code that makes the yaml file.

acossio, 2025-05-07:
> Yes! I agree this is a very important thing to do. Because one device can have multiple versions, I'll need to add a new table, eg device_firmware. This way, we can have multiple entries for one [...]

I think firmware is probably all we need to worry about. I'll double check with the PAM people but for our other sensors, firmware should be all that matters. 
With other changes over time, I'm wondering if we need a hardware version as well for some instruments. For example, we are the 4th or 5th version of the Nortek minis (titanium hull or non-titanium hull for example). The shadowgraphs have also changed hardware (small to large) as well as configuration (removal of a red-light filter). For the shadowgraphs, we might be able to make a note in the database or upload a document that would be a "calibration" to note that it had undergone a change. Most of the time, the Nortek's have come back with a different serial number.

smwoodman, 2025-05-07:
I forgot about the Nortek serial number thing. Maybe then this table would have a serial number too, so that those 'new devices' could all be lumped together. I'll think on this

smwoodman, 2025-05-22:
Created a new table Device_Firmware in the Access database. Per email discussion, it just has columns for 'Firmware version' and 'Firmware start date'. We can add other columns if we want.

Now to @acossio to enter firmware info, and find the inevitable bugs in the Access files