SWFSC/glider-lab #91: Investigate CTD pressure readings from amlr03-20231128

smwoodman, 2025-05-07:
## Background

Tony and Sam have been using amlr03-20231128 to test doing the Echoview acoustic processing with all Python (esdglider) outputs. During this process, Tony discovered that the depth values from the esdglider output were quite different than the Matlab data. We realized that when doing the processing with data from SOCIB, Tony was using the glider measured depth(i.e., m_depth), while the data from esdglider used depth calculated from the CTD's pressure sensor (i.e., depth_ctd). We confirmed that the way pyglider calculated depth is using the gsw package's z_from_p function, which is industry standard.

Normally these two values are only different by ~2 meters, because the instruments are in different places on the glider, and thus are at different depths when the glider is pitched up or down. Also, there can be a little bit of offset from drift in the times recorded by the flight and science computers. However, again the differences between these depths should be ~2 meters at most.

For amlr03-20231128, the difference in the depths from these two sources was up to 30 meters (i.e., m_dpeth was 30 meters greater than depth_ctd). Upon investigation, the depth_ctd value started the same as m_depth, but then drifted more and more . At ~200m there was a ~10m gap, and at ~900m there was a ~30m gap. This gap closed on the glider's climb, meaning that when the glider reached 200m again while climbing, depth_ctd was only 10m less than m_depth. This was consistent across the couple of profiles that we spot-checked. This consistent gap expansion and contraction means this difference is not the result of any drift, and instead seems to be a difference in the performance of the glider depth sensor and CTD pressure sensor.

We spot-checked a couple of other deployments: calanus-20241019, and amlr08-20220513. neither exhibited the same pattern (eg, amlr08-20220513 had a max depth difference of 2m across the whole deployment). This seems to further suggest the pattern seen in amlr03-20231128 is due to something with these specific sensors, rather than something across glider depth sensors and CTD pressure sensors as a whole.

Other notes: 
- Christian identified that the conductivity from amlr03-20231128 was bad, i.e. the conductivity cell was damaged somehow. Perhaps whatever damaged this also damaged the pressure sensor. 
- The CTD used on amlr03-20231128 (serial 9483) was last calibrated in 2018. amlr03-20231128 was it's third major (i.e., FREEBYRD) deployment.

## Next steps

- [ ] Tony will post a question to the UG2 slack channel
- [ ] The CTD has already been sent back for service. Tony will follow up with Seabird to see if they have ideas, and/or see anything during service
- [ ] Sam will process a few other deployments, and spot-check the difference in m_depth and depth_ctd. Deployments: amlr04 from the same 2023/24 season, one of the deployments from 2018 (when the CTDs were newest), and other deployments with CTD 9483.
- [ ]  Sam to update esdglider to not over-interpolate depth_ctd in the raw timeseries, and to sanity check compare m_depth and depth_ctd: https://github.com/SWFSC/glider-utils/issues/26
- [ ] Spot check depth and depth_ctd in a couple of SOCIB-processed timeseries

acossio, 2025-05-07:
I posted on the UG2 Slack under the data channel [here](https://app.slack.com/client/T02L6MX0BR8/C03JJ7S57JQ). I probably should have just put it under the general channel. First response wasn't very helpful.

acossio, 2025-05-12:
More responses from UG2 came in. Most point to a probable issue with the CTD, which is what we are leaning towards. I will find out who to write to at Seabird next.