SWFSC/glider-lab #100: Update metadata for unit_1024, risso, and stenella 2025 deployments

smwoodman, 2025-05-28:
@acossio @jgwalsh Could one or both of you please look at and update the metadata files for the spring 2025 deployments of unit_1024, risso, and stenella [here](https://github.com/SWFSC/glider-lab/tree/main/deployment-configs)? Items on my mind:
- update the comment and summary fields. Hopefully these can come right from the post-deployment reports that Jen has written?
- Confirm the 'contributor_name' for the various deployments
- Sanity check the glider_devices section (Tony)

Feel free to either commit changes directly to the repo, or email them to me. Once I get these from y'all, I'll update the processed data

acossio, 2025-05-29:
Thanks for doing this Sam. Here are a few comments/questions.
- For the creator name/email, should we be using Christian and his personal email? You should probably speak to him about it first, if you haven't already. We'll have to figure out who it should be from now on though. 
- Why is Seaglider in the keywords?
-  Did you fix the profile_time_start and profile_time_end from being output as Lattitude? I know this isn't part of the yml.
-  I added Renea Briner as a pilot for risso and stenella and pushed the changes
- The glider devices look good to me. 
- @jgwalsh can add the summary and comment fields from the reports she has been working on.

smwoodman, 2025-06-02:
> * For the creator name/email, should we be using Christian and his personal email? You should probably speak to him about it first, if you haven't already. We'll have to figure out who it should be from now on though.

Good shout. I've updated this email to nmfs.swfsc.esd-gliders@noaa.gov, since this is probably the mostly likely to be stable long-term. 

As a side-bar, at some point we'll need to bite the bullet and remove Christian from the 'creator' role in these metadata files. @acossio I'm thinking that we put your name there, and then we'll be covering our bases as best we can with you listed there and me listed under the publisher?

smwoodman, 2025-06-02:
> * Why is Seaglider in the keywords?

I know we just talked about this, but for posterity. Those keywords are something that I pulled directly from IOOS/Rutgers, so I'm just going to leave them as-is

acossio, 2025-06-02:
> > * For the creator name/email, should we be using Christian and his personal email? You should probably speak to him about it first, if you haven't already. We'll have to figure out who it should be from now on though.
> 
> Good shout. I've updated this email to [nmfs.swfsc.esd-gliders@noaa.gov](mailto:nmfs.swfsc.esd-gliders@noaa.gov), since this is probably the mostly likely to be stable long-term.
> 
> As a side-bar, at some point we'll need to bite the bullet and remove Christian from the 'creator' role in these metadata files. [@acossio](https://github.com/acossio) I'm thinking that we put your name there, and then we'll be covering our bases as best we can with you listed there and me listed under the publisher?

I'm fine if you want to have me as the creator for the metadata. Thanks.

closed 2025-07-29 by smwoodman