SWFSC/glider-lab #117: Things to do after the container

smwoodman, 2025-07-11:
Current glider team efforts are focused on prepping gliders and moorings to ship to the Antarctic. This issue is a list of housecleaning items to work on, after the container has been shipped in early to mid August. 

- #115 
- Sit in the same room, and make sure all necessary files are uploaded to GCP
- Sit in the same room, and make Database updates so that the Fleet Status Deployments page can be replaced by a code-driven script

smwoodman, 2025-07-28:
> * Sit in the same room, and make Database updates so that the Fleet Status Deployments page can be replaced by a code-driven scrip

See Tony's email titled "Some database ideas"

acossio, 2025-08-07:
- [ ]  clean up field laptops of data and excess files
- [ ] Update ADCP calibrations in database
- [ ] Update acoustic calibrations in database
- [ ] build Mooring website/repo
- [ ] Add acoustic calibration files (*.ecs) to glider repo
- [ ] Find or ask Seabird for old CTD calibrations of mooring CTDs
- [ ] Compile all mooring sensor documents (ADCPs, CTDs, Releases, beacons)

Reference of email titled "Some database ideas'
1. In order to cut down on duplicates, can a report from the database be created that simulates [Fleet Status](https://docs.google.com/spreadsheets/d/1tB3QNKYx8qOYYS9QZotekBAx0y-_n2d-EZPjFFdYNuU/edit?usp=sharing)? Basically, the glider build of the most recent glider. Maybe we make sure that the most recent glider is checkboxed as active? I'm not sure how to do this.
 2. We are also tracking service in Fleet Status. I'm thinking that we can add the RMA sheet of fleet staus as a report. We could use the Glider build as a way to track the service (a mini build if you will) and have a Project called Service. Then we could report open services or all Services. We would just use the start and end date as when we shipped the gear and when we received it.